adafruit/Adafruit_SH110x #2: Documentation copy/paste error

benzvan, 2021-01-22:
- Arduino board:  Particle Photon

- Arduino IDE version (found in Arduino -> About Arduino menu): vscode/particle workbench

- List the steps to reproduce the problem below 

In the README.md of this project I see:

>Preferred installation method is to use the Arduino IDE Library Manager. To download the source from Github instead, click "Clone or download" above, then "Download ZIP." After uncompressing, rename the resulting folder Adafruit_SSD1306. Check that the Adafruit_SSD1306 folder contains Adafruit_SSD1306.cpp and Adafruit_SSD1306.h.

But I think that is a holdover from the SSD1306 project and not correct for this SH110x project

ladyada, 2021-01-22:
could be! can you submit a PR with documentation changes?

benzvan, 2021-01-23:
I'd be more than happy to. Should I just `s/SSD1306/SH110X/g` in the README?

I haven't been able to get it to work on my Photon since it's so different. I happened to notice this when I was digging into it.

ladyada, 2021-01-24:
yep :)

benzvan, 2021-01-25:
https://github.com/adafruit/Adafruit_SH110x/pull/3

closed 2021-01-25 by ladyada