adafruit/Adafruit_SH110x #7: No example for SPI only i2c, SPI formatting

koogar, 2021-09-11:
- Arduino board:  QT-PY ATSAMD
- Arduino IDE version: 18.15

Hi,

I am struggle to get SPi working for the SH1106 with this library as there is only an example for i2c.
The screen is working with other SPi libraries using a 32u4 . 
It also works with i2c on this library with the QT-PY

The library appears to support SPi functions, as I took most of my test code from it. I think I just need to use the correct formatting. 

This is why I think an included Arduino example using SPi would be great for others :)

I am currently trying the below with no success. These pins work with an SPi TFT on the QT-PY

#define OLED_MOSI     10
#define OLED_CLK        8
#define OLED_DC         7
#define OLED_CS          5
#define OLED_RESET     9

// Create the OLED display
Adafruit_SH1106G display = Adafruit_SH1106G(128, 64, &SPI, OLED_DC, OLED_RESET, OLED_CS); // ???

I am also not sure what to use in the "setup" section to start the display in SPi

// Start OLED

display.begin(0, true); // we don't use the i2c address but we will reset! // ???

Thanks for you help and work on the library

Rupert

ladyada, 2021-09-12:
here is the macropad example https://learn.adafruit.com/adafruit-macropad-rp2040/arduino#example-code-3099272-13

closed 2021-09-12 by ladyada

koogar, 2021-09-12:
Thanks for the reply but that was the example I was using.
I delved in the library a bit more and got it working with the below.

Changed to
// Create the OLED display
Adafruit_SH1106G display = Adafruit_SH1106G(128, 64, OLED_MOSI, OLED_CLK, OLED_DC, OLED_RST , OLED_CS);

Thanks

Rupert