adafruit/Adafruit_SH110x #10: Adafruit_SH1106G not working on SPI

gps79, 2022-02-09:
- Arduino board:  **NodeMCU v3**
- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.15**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**
1. Build and run example https://github.com/adafruit/Adafruit_SH110x/blob/master/examples/OLED_QTPY_SH1106/SH1106_128x64_SPi_QTPY/SH1106_128x64_SPi_QTPY.ino

**Expected result:**
Software runs without errors.

**Actual result:**
Software crashes.

**Root cause:**
- file: Adafruit_SH110X.cpp
- method: void Adafruit_SH110X::display(void)
- line: uint16_t maxbuff = i2c_dev->maxBufferSize() - 1;
- cause: i2c_dev is null for SPI connection

ladyada, 2022-02-09:
are you using adafruit hardware? if so please post a photo!

gps79, 2022-02-09:
I don't use adafruit hardware.
I just pointed an error (i2c_dev is a null pointer and causes crash).
The fix for that is to move line:
uint16_t maxbuff = i2c_dev->maxBufferSize() - 1;
below line:
    if (i2c_dev) { // I2C

so it's executed only when  i2c_dev structure is initialized.
Then the SW doesn't crash.

caternuson, 2022-02-09:
What code are you running? Is it calling `begin()` on the display instance?

caternuson, 2022-02-09:
@gps79 If you can, please test the changes in PR #11.

closed 2022-02-09 by caternuson