adafruit/Adafruit_SH110x #16: display.clearDisplay() not clear properly on button pressed on ESP32

waq77, 2022-10-13:
hello all,
I am working with ESP32-TTGO-T1 board and PlatformIO IDE, I am clearing display (display.clearDisplay()) on button pressed and display another text in while(1) loop.
Whenever I pressed button, Old text not completely cleared, and new text overlapped everytime. 
kindly help, Thanks in advance

caternuson, 2022-10-13:
Please post complete code for an example sketch that demonstrates the issue.

waq77, 2022-10-14:
whenever I pressed button to move up and down , previous text not clear completely and new one written pn the top of it. 

while ((gpio_get_level(Select_Channel_Button) == 0 || pressed==false) && choice==true)
   {
      if (gpio_get_level(Channel_Decrement_Button) == 1 || (gpio_get_level(Channel_Increment_Button) == 1))
      {
       up=up+1;
       if(up>2)
       {
         up=1;
       }
     
       if (up==1)
       { 
         back=false;
         pressed=true;
         initial_val=false;
         chnl_sel=true;
         display.clearDisplay(); 
         display.setTextSize(1);
         display.setTextColor(SH110X_WHITE);
         display.setCursor(20, 3);
         display.println("Configurations");
         display.drawLine (0,13,128,13, SH110X_WHITE);
         display.setCursor(0, 50);
         display.println("Back");
         display.setTextColor(SH110X_BLACK, SH110X_WHITE);
         display.setCursor(0, 35);
         display.println(">Select_Channel");    
         display.display();
         vTaskDelay(pdMS_TO_TICKS(500)); 
        }
     
        if (up==2)
        {
          back=true;
          pressed=true;
          initial_val=false;
          chnl_sel=false;
          display.clearDisplay();
          display.setTextSize(1);
          display.setTextColor(SH110X_WHITE);
          display.setCursor(20, 3);
          display.println("Configurations");
          display.drawLine (0,13,128,13, SH110X_WHITE);
          display.setCursor(0, 35);
          display.println("Select_Channel");
          display.setTextColor(SH110X_BLACK, SH110X_WHITE);        
          display.setCursor(0, 50);
          display.println(">Back");
          display.display();
          vTaskDelay(pdMS_TO_TICKS(500)); 
        }
     }    
      vTaskDelay(pdMS_TO_TICKS(200));         
    }

waq77, 2022-10-14:
Thanks, my problem is solved. This problem occurs when will you are using display fuctions within multiple created tasks

closed 2022-10-14 by waq77