adafruit/Adafruit_SH110x #17: Adafruit 128x64 OLD Featherwing works on CircuitPython but not on BLINKA MCP2221

markbirss, 2023-02-27:
Connecting the same SH110X display using same library same example

Example:
https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SH1107/blob/main/examples/displayio_sh1107_simpletest.py

From:
https://learn.adafruit.com/adafruit-128x64-oled-featherwing/circuitpython

ESP32 CircuitPython
works as expected same screen output produced
https://learn.adafruit.com/assets/95402

BLINKA MCP2221 
same example produces garbled display

[image]

caternuson, 2023-02-27:
FeatherWings are meant to be used with Feather main boards.

Please post in the forums with photos of your setup showing how everything is connected:
https://forums.adafruit.com/

markbirss, 2023-03-01:
@caternuson 

Will post on the forum

It was connected by  Stemma QT with reset from MCCP221a to FeatchWingOLED

closed 2023-03-01 by markbirss