adafruit/Adafruit_SH110x #22: 128x128 SH1107 Flickering

christiankuhlmann, 2024-07-14:
### Adafruit SH1107 128x128 board flickering.
When using the adafruit 1.12" OLED board with this library flickering is observed. I've checked the library and can't see any obvious issues. Other threads have covered **SH110X_SETMULTIPLEX** being a potential issue yet with **0x3F** and **0x7F** the display is still seen to flicker. I would expect this issue is being caused by the SH1107 having a frame-rate rather than holding the pixels constant. I also tried playing around with **SH110X_SETDISPLAYCLOCKDIV** to no avail.

I don't really care that the refreshing is occurring and is visible by a camera. What I do care about is the strange effect created by physically moving the device.

Two questions:
Is it possible to disable the constant refreshing (assumedly caused by the SH1107 itself)?
Is this an issue with the library or an inherent issue with how the SH1107 works?
Any other solutions?

Video of issue:
https://github.com/user-attachments/assets/db479ffc-86a6-4824-91d1-2b669e004c5d

Jonathan-1of1, 2025-02-07:
Hi, christiankuhlmann
I just started working with the same display and am having the same issue - it makes scanning a QR code impossible.
Did you ever resolve the flickering?

roseundy, 2025-03-15:
I found an article [here](https://done.land/components/humaninterface/display/oled/sh1107/)
It said that the u8g2 code for the SH1107 didn't suffer from the flickering that the Adafruit library did.

Looking at the u8g2 code vs. the Adafruit code, the SETDISPLAYCLOCKDIV command uses a different value between the two. Adafruit uses a value of 0x51, while u8g2 uses 0x50.

When I hacked my own copy of the driver, change to 0x50 fixed the flickering issue for me.